KentuckySolarCar/RaspberryPi #4: General instability

cheintz, 2015-01-01:
Motor controller log stops after approximately 4659 samples and before batteries (N=1)

jebr224, 2015-01-02:
I advise you to read "How to Write a Useful Bug Report" https://issues.apache.org/bugwritinghelp.html
and all of https://issues.apache.org/bugwritinghelp.html.
Those documents will explain the type of information that is helpful when debugging problems. They will also explain how to present this information in a useful way. 

For example python always gives a error message when programs crash. Typicall the error messages will even include a line number (how nice) . Having this error message, would change how we view this issue. The startup script will log error messages in their own file(how nice). Next time just attach the python error message.

Thank you
John Broadbent

cheintz, 2015-01-02:
Those documents seem to apply to bugs that occur under carefully replicable scenarios. This is not necessarily the case with an embedded logging program operating on a vehicle traveling down the road and interfacing with rather sketchy external peripherals. I cannot reproduce errors on an offline system that is currently approximately 330 miles away from me.

Any error messages should be present in the log  file created by your run script(as you mentioned). I do not know the location of those logs. They were never pushed to the SVN. It is also conceivable that these logs do not contain any errors if the problem was external to the python code itself (motor controller stopped sending CAN, etc).

I did neglect to include the time and date of the log in which this was evident. The closest I can find as of now is both logs on 2014-12-13, where the motor controller stops being logged approximately one minute before the battery box stops.

It is also worth noting that this discrepancy may exist because the motor thread has been reported to induce a delay of tens of seconds for unknown reasons. It is reported that this was not an issue before the rewiring of the  car.

closed 2015-10-03 by egtoneyUK