KentuckySolarCar/RaspberryPi #5: Battery log data rate

cheintz, 2015-01-01:
Log contains 5hz data, but data is sent by battery box (and changes in the log at approximately 1hz)

jebr224, 2015-01-12:
https://github.com/KentuckySolarCar/RaspberryPi/commit/6ba57aed26257f5a02d566584c485c283e021d26

cheintz, 2015-01-18:
Battery data likely still only contains data from one pack.
Battery data still does not contain header (which should have units at some point)

cheintz, 2015-04-23:
As of the log 2015-04-07\00-11-21\battery.txt, these issues still exist.

crba233, 2015-04-25:
-Changed data log rate for batteries to 1 hz
-Battery header information added (solving the issue of only one battery pack appearing to be logged)

cheintz, 2015-04-30:
As of logs from 2015-04-30, battery header information is not present and lack of variation in data suggests that only one pack is logged.

Furthermore, data is still written to the file with timestamps approximately every 0.2 seconds, even though it only changes about every 2 seconds.

crba233, 2015-05-01:
This is fixed in the new code that has not been uploaded to the PI (unless Ethan did that today)

closed 2015-10-03 by egtoneyUK