KentuckySolarCar/RaspberryPi #6: Lack of documentation on startup script

cheintz, 2015-01-01:
Document how to start and stop the autorun script via SSH.

jebr224, 2015-01-02:
Hmmmm,

What level of documentation do think is required for this script?
I think most CS majors would understand that SolarCarStartUp.sh, controls the startup of program. Then they would run the script, and it would tell them "Usage:  {start|stop|restart|reload|condrestart|status}".
This is a normal way Linux people stop and start services. 

Maybe I presume to much? For you, Chris darling, I will spell it out.

The command below will stop all instances of python running on the pi
sudo SolarCarStartUp.sh stop

The command below will start the logging script
sudo SolarCarStartUp.sh start

If you genuinely think that is not obvious, I encourage you to write up a clear and short document explaining the script.

Thank you
John Broadbent

cheintz, 2015-01-02:
John,

I created this issue because Ethan expressed that he did not know how to stop the automatically started python instances, which resulted in his inability to fix the code. This has led to lost/missing log data, which inhibits my ability to prioritize improvements to the mechanical systems of the car.

Thank you for documenting in writing the usage of the script, as well as your tendency to talk down to people from your Linux high horse. I am truly happy for you that you can derive happiness from such unproductive and childish acts.

closed 2015-01-02 by cheintz