KentuckySolarCar/RaspberryPi #7: Motor controller log on~1hz

cheintz, 2015-01-02:
Data is logged at ~5hz, but only actually changes at ~1hz. CAN messages are sent by the motor controller at 5hz for at least some of the messages. Vehicle speed, bus current, and phase current are the most important to have at this data rate.

This has been a problem since on or before 2014-10-24

jebr224, 2015-01-12:
https://github.com/KentuckySolarCar/RaspberryPi/commit/d8e9e74241118bdef71ef2e5634f31198f90425f

cheintz, 2015-04-23:
As of the logs 2015-04-07\00-11-21\motor.txt, the data rate issues still exists.

cheintz, 2015-04-30:
As of logs from 2015-04-30, the data is timestamped every 0.2 seconds (desired), but does only changes approximately every 0.6 - 0.8 seconds (3-4 samples)