KentuckySolarCar/RaspberryPi #16: Parse Arduino Data

egtoneyUK, 2015-09-15:
All data from the Arduino needs to be parsed and written to the respective database table. This has already been partially implemented in the [ArduinoSerialIO.py](https://github.com/KentuckySolarCar/RaspberryPi/blob/master/SerialProg/ArduinoSerialIO.py) file.

ArthurSilveira, 2015-09-19:
On it!

ArthurSilveira, 2015-09-19:
I'm running into a problem with parsing and matching the data from our GPS.

The data from the GPS doesn't match what we should be getting from it.

**Sample data from our GPS:** 
$GPRMC,000051.400,V,,,,,**0.00**,0.00,060180,,,**N*42**

**What the data should look like according to the documentation**
$GPRMC,081836,A,3751.65,S,14507.36,**E**,000.0,360.0,130998,011.3,**E*62**

The empty parts of the string aren't a problem, however 2 big issues are present above. 
1. At the end of the string, we should get an **East(E) or West(W)** value and we are getting an **N**
2. The 6th location from the end should return **East(E) or West(W)**  as well and instead we are getting a number.

Can the GPS be compromised?
If anyone can shed some light on this it would be awesome.

cheintz, 2015-09-20:
It is possible that the GPS does not fully follow the standard and instead
applies the N and E deisgnations with numbers, or a sign on the actual
coordinates.

It is also possible that the arduino (that sits between the GPS receiver
and the PI) is subtly changing the NMEA sentences.

weiliansong, 2015-09-20:
Can you find the driver for GPS on the arduino? Maybe it will tell you how its ouxtputting the data.

dcambron, 2015-09-20:
I believe we use this:

https://github.com/KentuckySolarCar/UKSC_Arduino_Libraries/tree/master/libraries/Adafruit_GPS

ArthurSilveira, 2015-09-20:
Daniel, this looks right thanks.

weiliansong, 2015-09-20:
Also it seems like the 6th value is just blank, not 0.00. Or maybe i just cant count commas. Can anyone double check on that?

ArthurSilveira, 2015-09-20:
6th starting at the end going backwards :D

weiliansong, 2015-09-20:
Oh i guess i can count, but not read.... Will help to look at the arduino file soon

weiliansong, 2015-09-20:
Im pretty sure you already knows this, but there are two types of GPRMC sentences, one with 12 fields and another with 13 fields. See if this website will help you. 

http://playground.arduino.cc/Tutorials/GPS 

Right before the arduino code is where it talks about the $GPRMC fields

closed 2015-09-29 by ArthurSilveira