KentuckySolarCar/RaspberryPi #17: UDEV Rules to set Unique Serial Ports

egtoneyUK, 2015-09-15:
This entails writing UDEV rules that assign serial connection to a unique port. For example the batteries should be ttyBAT#. This will also include messing with the FTDI chips to somehow assign a unique value.

weiliansong, 2015-09-19:
The formatting of devices is "tty" + Device Name("MOT" for motor, "TEL" for telemetry, "BAT" for battery) + Number of directory(Ex: "3" of sda3).

Ex: ttyMOT1, ttyBAT0, ttyBAT2, ttyTEL3
              names above correlates to:
      ttyUSB1,  ttyUSB0, ttyUSB2, ttyUSB3

egtoneyUK, 2015-09-19:
Okay, this has been completely implemented on the Python side! More testing needs to happen but it's complete.

egtoneyUK, 2015-09-27:
The Telemetry rule is not working. No symbolic link, only a /dev/ttyUSB* appears for it.

weiliansong, 2015-09-27:
Seems like one of the Match Keys in udev rule is not correct, will have to check the value for each attribute for the telemetry device and change the udev rule accordingly so it can identify the right device. Are you by chance near the car?

weiliansong, 2015-09-27:
Oh never mind I found the issue. One of the word in udev rule is mis-spelled, correcting the error as we speak

closed 2015-10-03 by egtoneyUK